IObundle/iob-soc #45: Bigger bin files transfer blocks the console

AndreMerendeira, 2020-07-27:
When transfering bin files bigger than the one generated by the Hello World, the console blocks because it waits for the bootloader response. This is because the bootloader seems to block while receiving  the program and writing it to mem. Both program sizes are correct, but the for loop stops while retrieving the program sent through the UART.
Maybe there is some data loss along the transfer that the write() function doesn't detect

AndreMerendeira, 2020-07-27:
After some debuging, the problem seems to be caused by the write() not being configured as blocking. Sending one byte at a time and waiting after every 200bytes that were sent seems to solve the problem.
Not a very good and stable solution, needs developing.

FridayFawkes, 2020-07-28:
When the program is sent in a single block, the receiving side can't keep up with the data transfer and the buffer on the receive side, for larger .bin files ends up being filled up / overflown, causing the loss of data. 
In those cases the bootloader can't read the expected amount in file_size and keeps waiting on uart_getc().
A more stable solution would involve a more robust protocol by segmenting the .bin file into smaller chunks and waiting from an ACK from bootloader between each sent chunck. Or implementing another existing simple file transfer protocol.

jjts, 2020-07-28:
I thought  the problem was on the PC side: the comm port not properly configured as blocking....

But if this is a bootloader problem it should be easy to solve by checking the UART hardware and software.

At 115200 baudrate a byte takes an eternity to arrive for a CPU that works at 100MHz. Maybe what is happening is that the UART cannot receive the next byte before the bootloader removes the previous byte. Can the bootloader store the previous byte while the current is being received?

jjts, 2020-08-27:
solved by sending bytes one by one

closed 2020-08-27 by jjts