IObundle/iob-soc #146: Python2 deprecated

AndreMerendeira, 2021-07-08:
Python2 is no longer distributed in more recent OS's, so installing the toolchain pre-requisites gives out an error `E: Unable to locate package python2`. In the more recent toolchain versions, python2 has been removed from the pre-requisites installation command. Solution: update Readme file

JDLopes, 2021-07-08:
> Python2 is no longer distributed in more recent OS's, so installing the toolchain pre-requisites gives out an error `E: Unable to locate package python2`. In the more recent toolchain versions, python2 has been removed from the pre-requisites installation command. Solution: update Readme file

Python2 is not, but Python version 2.7.x still works. You can run 'python --version' in a new Linux installation (at least in Ubuntu variants).

AndreMerendeira, 2021-07-08:
So, only remove python2 from the pre-requisites install and keep the linkage the same?

JDLopes, 2021-07-08:
> So, only remove python2 from the pre-requisites install and keep the linkage the same?

I see your point, but since it is part of the riscv-gnu-toolchain installation flow, we should keep it as the authors have on their readme.

AndreMerendeira, 2021-07-08:
That's the thing, they changed it, no more need for python2

JDLopes, 2021-07-08:
> That's the thing, they changed it, no more need for python2

So we need to update our readme. I missed that, sorry.

P-Miranda, 2021-11-11:
I have updated the iob-soc `software/python/` scripts to python3
Is it possible to create a `146-python3` branch so I can PR to upstream and perform Jenkins verification?

jjts, 2021-12-15:
can I close this one

JDLopes, 2022-01-12:
> can I close this one

Yes, you can.

closed 2022-01-12 by jjts