IObundle/iob-soc #173: fpga-build target fails

P-Miranda, 2021-09-07:
### How to reproduce:
- on a clean iob-soc clone run:
```
make fpga-build
```
### Error: 
```
make[1]: *** No rule to make target `../../../../software/bootloader/boot.bin', needed by `boot.hex'.  Stop.
```
### Problem:
```
make fpga-build
```
1. runs `build` target in `fpga.mk`, which requires `$(FPGA_OBJ)`.
2. `$(FPGA_OBJ)` requires `boot.hex` (target in `hardware.mk`).
3. `boot.hex` requires `$(BOOT_DIR)/boot.bin`, for which there is no target. The `boot.bin` file is only generated when the `sw` target is required (for example in the case of `make fpga-run`).
### Proposed solution
Create a target for `$(BOOT_DIR)/boot.bin` 
(optional) use boot.bin as requisite for the `sw` target:
```
[hardware.mk file]
boot.hex: $(BOOT_DIR)/boot.bin                                                  
    $(PYTHON_DIR)/makehex.py $(BOOT_DIR)/boot.bin $(BOOTROM_ADDR_W) > boot.hex  

$(BOOT_DIR)/boot.bin:
    make -C $(BOOT_DIR) boot.elf FREQ=$(FREQ) BAUD=$(BAUD) 

# make embedded sw software                                                     
sw: $(BOOT_DIR)/boot.bin                                                                             
    make -C $(FIRM_DIR) firmware.elf FREQ=$(FREQ) BAUD=$(BAUD)
    make -C $(BOOT_DIR) boot.elf FREQ=$(FREQ) BAUD=$(BAUD)                  
    make -C $(CONSOLE_DIR) INIT_MEM=$(INIT_MEM) 
```

jjts, 2021-09-07:
seems to be fixed by last commit

jjts, 2021-09-07:
if yes please close issue

P-Miranda, 2021-09-07:
Issue fixed

closed 2021-09-07 by P-Miranda