IObundle/iob-soc #229: make sram does not work

jjts, 2021-11-03:
In branch asic 

make sram 

needs $OPENRAM_HOME/myconfig.py which is not present 

This is a user created file: it should not be placed in $OPENRAM_HOME/

How to solve 

1) write a python script and place it in the iob-mem repo -- not iob-soc 

this script should create the openram config file . for example 

python3 mkopenram_config.py TYPE WORDS WIDTH 

then it creates openram_config.py 

In the Makefile modify the recipe to read this file instead of what you have there

<sram model file>:   openram_config.py 
     python3 $(OPENRAM_COMPILER)/openram.py $<

openram.py:
   $(MEM_DIR)/software/python/mkopenram_config.py <TYPE> $(SRAM_ADDR_W) $(DATA_W)

Or something like that

jjts, 2021-12-15:
can this one be closed too?

microSharjeel, 2021-12-15:
Yes it can be closed

closed 2021-12-15 by jjts