IObundle/iob-soc #282: Problems in compilation of iob-soc in yosys

microSharjeel, 2021-12-14:
Hi Sir,
Kindly, do 
make asic in hardware/asic/skywaterMakefile
and there is a compilation error given by the yosys which is as follows,
12. Executing Verilog-2005 frontend: src/uart_core.v
ERROR: No such module: system
You can reproduce it by make asic inside the skywater folder
When you will reproduce this we will discuss this. I just want to elaborate that "  system " is the name of the top level module. OpenLane requires this in order to synthesize the design.
regards

jjts, 2021-12-14:
Maybe this is because there are some System Verilog in uart and other cores.

Is there a system Verilog switch for yosys ?

microSharjeel, 2021-12-14:
Sir, I think this may not be the case

microSharjeel, 2021-12-15:
I have updated the issue and now the real problem is that VSRC and VHDR do not have all the source and header

JDLopes, 2021-12-15:
It is fixed!

jjts, 2021-12-15:
Hi Sharjeel 

Please confirm it is fixed so I can close this

Also please study Joao's fix for the next time I hope you can do it yourself

microSharjeel, 2021-12-15:
Hi 
This issue is now resolved  and can be closed 
regards

closed 2021-12-15 by jjts