IObundle/iob-soc #288: Guidance on replacing Picorv32 Core with another

TheCopperMind, 2022-01-03:
Hi,

I am currently working on a custom RISC-V core and came across this project, and I was wondering if there exists/ are any plans for guidance on how to swap out the existing CPU with a different one?

TheCopperMind, 2022-01-03:
Thank you for the quick reply, the interface was actually the first question I have. I'll make sure to look at the interconnect. I was also wondering about the AXI-4 interface, that is in the Picorcv32, do I need to provide that as well?

jjts, 2022-01-03:
You are welcome. 

Although we use Verilog macros to avoid always writing the same thing our "native" interface is quite easy:

Request bus from master: valid, addr, wstrb (this is a 4-bit byte enable), write data, read data (in) and ready (in) //all others are outs

Response bus from slave: valid (in), read data (out) and ready (out)

You should just pay attention to how it works:

The master may assert valid any time. Ready from slave takes at least one cycle after valid (cannot come immediately and combinatorially after valid but can come in the next cycle)

The Valid signal may be asserted while ready is on, requesting the next transaction. If the slave can respond in the next cycle, we have a full pipeline with one word passed on each cycle from master to slave.

This interface has much fewer signals than AXI and is good enough for most of the things we do. However, to interface with 3rd party hardware we use the native-axi adaptors you can also find in the interconnect repo.

Enjoy.

TheCopperMind, 2022-01-06:
Hi, thanks for the in-depth explanation. Does this apply to both reads and writes?

jjts, 2022-01-06:
Yes, it does, but I forgot the write part on the slave:

full slave bus: valid (in), write data (in), 4-bit write strobe (in), read data (out) and ready (out)

TheCopperMind, 2022-01-11:
I have another question which I hope has an obvious answer that I haven't seen, but how do we make a request to both read an instruction and read from data memory within the same cycle?

jjts, 2022-01-11:
The data and instruction buses are separated, so nothing prevents you from doing so.

If using SRAM, we use dual-port memories to allow this.

Using DDR, we use two separate AXI ports of an external AXI-interconnect. We used Xilinx FPGAs' AXI interconnect IP and DDR controller generated by the MIG.

It happens that PicoRV32 has a single bus for both instructions and data. To make it compatible with IOb-SoC, we created the iob-picorv32 submodule, which demuxes this bus to provide two separate buses for instructions and data.

There's no need to demux for other CPUs that natively provide these two buses, there's no need to demux, and the wrapper can be more straightforward.

I hope this helps.

TheCopperMind, 2022-01-11:
Thank you for the swift response again! It's very helpful. I see how it works now.

closed 2022-04-29 by jjts