IObundle/iob-soc #399: stucks when i simulate with INIT_MEM=0 RUN_EXTMEM=1

zaliasc, 2022-05-14:
Hello Community,
I try to simulate the iob-soc project with the command (in root directory):
`make sim-run SIMULATOR=icarus INIT_MEM=0 RUN_EXTMEM=1`

but it stucks here:
[image]

I think this problem has something to do with uart，so i try to debug the python file: software/console/console.
I add some debug message and run it again:
[image]

the result is:
 
[image]

It seems IOB-console script do not receive END flag, and stucks at `tb_read()`
I'm not sure if some dependencies are missing.

Thanks in advance.

distribution:  Ubuntu 18.04

jjts, 2022-05-14:
Hi @zaliasc, did you make sim-clean before running that? It may have previously compiled objects. Also, note that Icarus is relatively slow, so wait for a little before concluding it is stuck

zaliasc, 2022-05-15:
Hi @jjts, thanks for your advice.
I will try it later again, but it's not necessary. I just try to use part of the iob-soc project, so i will close the issue.
The project is very informative, thanks again.

closed 2022-05-15 by zaliasc