IObundle/iob-soc #430: Verilog simulation error

jjts, 2022-07-24:
make sim-run SIMULATOR=xcelium RUN_EXTMEM=1 VCD=0 INIT_MEM=0

Last message:

Open failed on file "cnsl2soc". No such file or directory

PedroAntunes178, 2022-11-12:
The simulation ends when the `cnsl2soc` file is no longer available. Most simulators see that the file is unavailable and return 0 when trying to open it. `xcelium` finishes the simulation similar to the other simulators but prints an error when it cannot open the `cnsl2soc` file.

jjts, 2022-11-13:
Upon exiting a message telling the user to ignore that error should be issued. After that this issue can be closed.

jjts, 2022-11-14:
If the simulator doe snot give this error the suggested above message would be nonsensical

closed 2022-11-14 by jjts