IObundle/iob-soc #509: __pycache__

jjts, 2023-01-28:
left all over

arturum1, 2023-01-30:
Solved in commit https://github.com/arturum1/iob-lib/commit/b5e4829899924d75f41e445d495637af4b79eb62

PedroAntunes178, 2023-01-30:
I had an easier solution:
```
$(BUILD_DIR): python_setup python-cache-clean

python_setup:
	./$(SETUP_PYTHON_FILENAME) $(SETUP_ARGS)
```

arturum1, 2023-01-30:
That solution involves creating and deleting files while the one I proposed using the -B flag prevents creating the files altogether.
Also, that solution `$(BUILD_DIR): python_setup python-cache-clean` may not work well if python_setup fails midway because that will exit Make without calling python-cache-clean to clean the files.

jjts, 2023-01-30:
It is always better to prevent garbage than to garbage collect later

PedroAntunes178, 2023-02-02:
@jjts I believe this issue can be closed, since it was solved in the referenced PRs. (https://github.com/IObundle/iob-soc/pull/513 and https://github.com/IObundle/iob-lib/pull/416)

PR #513 (2023-01-31): Add setup arguments; FPGA board configuration at build-time; Remove 'meta' dictionary.
- Enable FPGA tests in ci.yml
- Create test.expected for each possible configuration of INIT_MEM and RUN_EXTMEM.
- Add sim_setup.py and fpga_setup.py to copy correct test.expected file to the build directory
- Allow selecting FPGA board at build-time. No longer configured at setup. Resolves #512
- Add INIT_MEM and RUN_EXTMEM setup arguments. Details here 9993feed44a846cfe02a58de67e62e4e7d516769.
- Updated TEST_LOG. Resolves #511
- Moved simulation sources. Resolves #506 
- Updated LIB. Resolves #509

Check c [... cut]
closed 2023-02-02 by jjts