IObundle/iob-soc #510: sw compilation flags

jjts, 2023-01-29:
what is this?

riscv64-unknown-elf-gcc -o iob_soc_firmware.elf -Os -nostdlib -march=rv32i -mabi=ilp32 --specs=nano.specs -Wcast-align=strict -Wl,-Bstatic,-T,../template.lds,--strip-debug -Wl,-Map,iob_soc_firmware.map -DINIT_MEM -DINIT_MEM -DINIT_MEM -DINIT_MEM -DINIT_MEM -I. -I.. -I../esrc -I../src -I../firmware  ../firmware/iob_soc_firmware.c ../firmware/iob_soc_firmware.S ../esrc/iob_cache_swreg_emb.c ../esrc/iob_uart_swreg_emb.c ../src/iob-cache.c ../src/iob-uart.c ../src/printf.c -lgcc -lc -lnosys

jjts, 2023-01-29:
-DINIT_MEM -DINIT_MEM -DINIT_MEM -DINIT_MEM -DINIT_MEM -I. -I.. (?)

PedroAntunes178, 2023-02-02:
The multiple appearances of -DINIT_MEM were happening because config_build.mk was being included multiple times in the Makefiles.
Now, since the INIT_MEM define has passed to the setup time, it no longer appears in the SW compilation arguments.

The includes `-I.` and `-I..` are necessary because there are files which need to be included in the bootloader and firmware compilation which are in the Makefile directory (`.`, it could be either software/pc-emul or software/embedded directory) and in the software directory which is `..`.

closed 2023-02-02 by jjts