IObundle/iob-soc #611: Vivado.log critical errors related to the Ethernet and the "clk" constrains

PedroAntunes178, 2023-10-28:
Hello,

### Issue Description:  
There are a few critical warnings which I believe could be solved relatively easily.

A lot of the critical warnings present on the vivado.log file are related to the Ethernet constrains. This happens when the Ethernet is unused. The constraints are in the file [iob_soc_fpga_wrapper_dev.sdc](https://github.com/IObundle/iob-soc/blob/main/hardware/fpga/vivado/AES-KU040-DB-G/iob_soc_fpga_wrapper_dev.sdc) which is always included.

Another alarming warning is:
```
CRITICAL WARNING: [Constraints 18-1055] Clock 'clk' completely overrides clock 'c0_sys_clk_clk_p', which is referenced by one or more other constraints. Any constraints that refer to the overridden clock will be ignored.
```
This is caused by the inclusion of the [iob_soc_fpga_wrapper.sdc](https://github.com/IObundle/iob-soc/blob/main/hardware/src/iob_soc_fpga_wrapper.sdc) file even when the DDR is in use. On the Kintex UltraScale that makes the `create_clock -name "clk" -period $clk_period [get_ports "$clk_port"]` override the DDR clock constrains that are automatically created.

### Solution proposal:  
Create a separate file for the Ethernet constrains. The constraints in [no_ddr.sdc](https://github.com/IObundle/iob-soc/blob/main/hardware/fpga/vivado/AES-KU040-DB-G/no_ddr.xdc) and in [ddr.sdc](https://github.com/IObundle/iob-soc/blob/f2d6bf536f2203e65ea870a2528e4565e5b8ffe6/hardware/fpga/vivado/AES-KU040-DB-G/ddr.xdc#L264) should also be removed.

Only add `create_clock -name "clk" -period $clk_period [get_ports "$clk_port"]` if USE_EXTMEM is not defined.

jjts, 2024-01-23:
Is this fixed ?

PedroAntunes178, 2024-01-29:
> Is this fixed ?

Hi professor, I do not think so.

jjts, 2024-08-17:
Please add command to reproduce

arturum1, 2024-08-19:
1) Currently, ethernet constrains are commented, so the warnings no longer appear. See:
[Ethernet contraints for ku040](https://github.com/IObundle/iob-soc/blob/if_gen2/hardware/fpga/vivado/AES-KU040-DB-G/iob_soc_fpga_wrapper_dev.sdc#L77-L124)
[Ethernet contraints for cyclonev](https://github.com/IObundle/iob-soc/blob/if_gen2/hardware/fpga/quartus/CYCLONEV-GT-DK/board.tcl#L50-L77)

2) The DDR `clk` constraint warning still appears when we build iob-soc with `USE_EXTMEM=1`:
```
make fpga-run BOARD=AES-KU040-DB-G INIT_MEM=0 USE_EXTMEM=1
```
```
CRITICAL WARNING: [Constraints 18-1056] Clock 'clk' completely overrides clock 'c0_sys_clk_clk_p_i'.
New: create_clock -period 4.000 -name clk [get_ports c0_sys_clk_clk_p_i], [/home/anobrega/artur/iob_soc_V0.7/hardware/fpga/vivado/AES-KU040-DB-G/iob_soc_fpga_wrapper_dev.sdc:2]
Previous: create_clock -period 4.000 [get_ports c0_sys_clk_clk_p_i], [/home/anobrega/artur/iob_soc_V0.7/hardware/fpga/ip/ddr4_0/ddr4_0/ddr4_0_in_context.xdc:1]
Resolution: Review the constraint files and remove the redundant clock definition(s). If the clock constraints are not saved in a file, you can first save the constraints to an XDC file and reload the design once the constraints have been corrected.
```

---

We could solve both issues by generating iob-soc contrains with py2 scripts based on the values of `USE_ETHERNET` and `USE_EXTMEM` python parameters.

closed 2024-12-09 by jjts