IObundle/iob-soc #621: mkregs should have an opetion to connect the insance ports to the modules' ports

jjts, 2023-11-05:
Right now, it connects to internal wires only. It should do both.

arturum1, 2024-08-14:
Currently [`csr_gen.py`](https://github.com/IObundle/py2hwsw/blob/main/py2hwsw/scripts/csr_gen.py#L426) (old mkregs) always connects the csrs to the modules' [default interface](https://github.com/IObundle/iob-soc/blob/if_gen2/lib/hardware/modules/iob_uart/iob_uart.py#L51) defined by the [`csr_if` attribute](https://github.com/IObundle/py2hwsw/blob/main/py2hwsw/scripts/iob_core.py#L85).

The current implementation does not allow connecting csrs to internal wires.

Maybe we should create a standard `csrs` block that can be added like any other to the module's `blocks` list.
This would allow the user to connect this block in any way that it sees fit.
It would also allow us to remove the `csr_if` attribute from iob_core.py, and instead make it a python parameter of this new `csrs` block.

For example, in the `blocks` list of a module, we could add the following block:
```
{
    "core_name": "csrs",  # Keyword to let py2hwsw know this is the csrs block
    "instance_name": "swreg_inst",  # Name of the generated verilog instance
    "instance_description": "Swreg instance",
    "csr_if": "iob", # Specify the type of interface used to control registers
    "connect": {
        # Connect clock and control interfaces to module's ports
        "clk_en_rst": "clk_en_rst",
        "iob": "iob",
        # Connect registers to internal module wires
        "reg1": "reg1_internal_wire",
        "reg2": "reg2_internal_wire",
    },
},

```

arturum1, 2024-08-14:
Another idea would be to make csrs independent of the py2hwsw scripts, and instead create a new lib module for csrs.
This new lib module, for example `csrs.py` would be configured via python parameters (like in [iob_split.py](https://github.com/IObundle/iob-soc/blob/if_gen2/lib/hardware/modules/iob_split/iob_split.py#L2-L7)), and could be added to the `blocks` list of a module, like any other submodule.

For example, in the blocks list of a module, we could add the following block and define its registers via python parameters:
```
{
    "core_name": "csrs",  # Name of the submodule we are using (The `csrs.py` core from lib)
    "instance_name": "swreg_inst",  # Name of the generated verilog instance
    "instance_description": "Swreg instance",
    "connect": {
        # Connect standard and control interfaces to module's ports
        "clk_en_rst": "clk_en_rst",
        "iob": "iob",
        # Connect registers to internal module wires
        "reg1": "reg1_internal_wire",
        "reg2": "reg2_internal_wire",
    },
    "csr_if": "iob",  # Specify the interface used to control registers
    "csrs": [
        {
            "name": "uart",
            "descr": "UART software accessible registers.",
            "regs": [
                {
                    "name": "SOFTRESET",
                    "type": "W",
                    "n_bits": 1,
                    "rst_val": 0,
                    "addr": 0,
                    "log2n_items": 0,
                    "autoreg": True,
                    "descr": "Soft reset.",
                },
                {
                    "name": "DIV",
                    "type": "W",
                    "n_bits": 16,
                    "rst_val": 0,
                    "addr": 2,
                    "log2n_items": 0,
                    "autoreg": True,
                    "descr": "Bit duration in system clock cycles.",
                },
                {
                    "name": "TXDATA",
                    "type": "W",
                    "n_bits": 8,
                    "rst_val": 0,
                    "addr": 4,
                    "log2n_items": 0,
                    "autoreg": False,
                    "descr": "TX data.",
                },
            ],
        },
    ],
},

```

Some advantages of this implementation:
- No need for py2hwsw to parse a special keyword `csrs` like I suggested in the previous comment.
- `csrs` module becomes consistent with other modules, having its own setup directory, sources, scripts, ...
- Simplifies py2hwsw codebase by removing all csrs/swregs scripts.

closed 2024-08-17 by jjts