IObundle/iob-soc #759: create_wrapper_files

jjts, 2024-02-15:
What wrapper what files? Isn't there a better name?

cyclonev related scripts should go to the cyclonev directory

do not shebang files that do not run standalone and do not give them x permission (fixed)

jjts, 2024-02-15:
Add a header comment explaining what the files does

arturum1, 2024-02-16:
For now, every iob-soc python script (except for the main iob_soc.py) is located in the `scripts/` directory.
Should I create new scripts in each of the boards/simulation directories with functions dedicated for them?

jjts, 2024-02-16:
yes, please

each board, simulator, etc has its own folder 

except synthesis as for now we only use one tool

jjts, 2024-02-16:
some  files belong to the FPGA compile tool, not to any specific board

arturum1, 2024-08-14:
Removed create_wrapper_files script in commit https://github.com/IObundle/iob-soc/pull/906/commits/834f533cad7e97e739bd43204514116c24f95cff.
Currenlty, in if_gen2, fpga and simulation wrappers are generated with py2hwsw.

closed 2024-08-14 by jjts