IObundle/iob-soc #761: Eliminate setup.mk

jjts, 2024-02-16:
call bootstrap.py directly like in branch if_gen

arturum1, 2024-08-14:
Bootstrap.py no longer exists on if_gen2.

Current if_gen2 [setup.mk](https://github.com/IObundle/iob-soc/blob/if_gen2/lib/setup.mk) file only contains Nix and clean targets. Should these targets be moved to every Makefile that uses them?
For example, should we copy the current contents of [setup.mk](https://github.com/IObundle/iob-soc/blob/if_gen2/lib/setup.mk) to the [iob-soc Makefile](https://github.com/IObundle/iob-soc/blob/if_gen2/Makefile#L21) and [iob-cache Makefile](https://github.com/IObundle/iob-cache/blob/if_gen2/Makefile#L10)?

jjts, 2024-08-14:
don't the comercial cores always run nix-shell and clean by themselves?

arturum1, 2024-08-14:
Currently, commercial cores also include the `setup.mk` to use its clean target.
They always run nix-shell, but I think we could make the open-source cores more accessible if we make nix-shell optional.

jjts, 2024-08-17:
After checking, the entire lib directory should be moved to py2hwsw. So this issue will be closed and another opened.

jjts, 2024-08-17:
After all, I decided to reopen this.

if [ -d "$(BUILD_DIR)" ]; then $(call IOB_NIX_ENV, py2hwsw $(CORE) clean --build_dir '$(BUILD_DIR)'); fi
	@rm -rf ../*.summary ../*.rpt 
	@find . -name \*~ -delete

IOB_NIX_ENV is not needed: call nix by hand 

py2hwsw should check whether BUILD_DIR exists , not the makefile

the other lines are not needed , we will find a way to do them

Delete this script

closed 2024-08-21 by jjts