IObundle/iob-soc #805: uart if

jjts, 2024-03-13:
https://github.com/IObundle/iob-soc/blob/cd83a441a2acf143f21e3c9c471b74e1a729f4a9/iob_soc.py#L219

if all uarts have sme ext if then move this to lib

arturum1, 2024-03-13:
That interface is created in iob_soc.py because we are defining the ports/IOs of the IOb-SoC system in that section.

However, if we update the `if_gen.py` script such that it can generate the `rs232` ports automatically, then we can delete these lines: https://github.com/IObundle/iob-soc/blob/cd83a441a2acf143f21e3c9c471b74e1a729f4a9/iob_soc.py#L226-L250

jjts, 2024-03-14:
if_gen is only science fiction at the moment

arturum1, 2024-08-14:
Current py2hwsw automatically generates signals for rs232 interface. Example in [iob_soc.py](https://github.com/IObundle/iob-soc/blob/if_gen2/iob_soc.py#L251-L253).