IObundle/iob-soc #814: IOb-SoC firmware start address and size.

arturum1, 2024-03-19:
Add method of configuring IOb-SoC firmware start address and size, to prevent collisions in shared external memory with other systems.
This change will deprecate the [`MEM_ADDR_OFFSET` parameter](https://github.com/IObundle/iob-soc/blob/main/iob_soc.py#L319-L326).
We will no longer need this hardware logic: https://github.com/IObundle/iob-soc/blob/main/hardware/src/iob_soc.v#L325-L326

Edw590, 2024-08-11:
How would the start address be configured? A macro in iob-soc.py? Or in some other way?

What do you mean with configure size though? What size? (Isn't it fixed by what is written on firmware.c?)

arturum1, 2024-08-14:
I think it would be best if the firmware address and size were configured via iob-soc python parameters like these ones: https://github.com/IObundle/iob-soc/blob/if_gen2/iob_soc.py#L11
It may also be required to use macros (from the 'confs' list) since they may be included in software sources.

By configurable size, I mean that the firmware stack pointer should not be placed at the [highest memory address](https://github.com/IObundle/iob-soc/blob/revamp2/software/src/iob_soc_firmware.S#L13) but instead at the end of the configured firmware size.

The end goal would be to have two (or more) parallel iob-soc systems sharing the same memory.
Each having its own firmware while avoiding collisions in memory.

jjts, 2024-08-14:
tend to agree -- revamp2 is in a middle of this storm and we are not even considering multiple CPUs

Edw590, 2024-08-21:
Done on revamp2

jjts, 2024-08-21:
not so fast, the tests are still failing, it has not been merged to if_gen2, let alone in the main branch.

closed 2024-09-10 by jjts