IObundle/iob-soc #948: iob_r input/output signals {iob_r_data_i/o} should be renamed do data_i/o

AndreMerendeira, 2024-09-18:


P-Miranda, 2024-09-18:
I think that the new name should be more explicit regarding the use of iob_r as a synchronization register.
We use the output signal name to match for vivado's `ASYNC REG`. If we use `data_i/o` for iob_r, the name could match other signals in the circuit.
Something like `sync_data_i/o` or `cdc_data_i/o`.

AndreMerendeira, 2024-09-18:
> I think that the new name should be more explicit regarding the use of iob_r as a synchronization register. We use the output signal name to match for vivado's `ASYNC REG`. If we use `data_i/o` for [...]

yes, you're right, didn't thought about that

jjts, 2024-09-18:
it is irrelevant iof used in synchronizer -- this is a basic register 

what I this issue really about ?

what branch ?

what was the old name ?

what is the new name ?

I will close this as unintelligible if not clarified

jjts, 2024-09-18:
Ok, now I understand, but perhaps there's more to this issue.

These modules are intended as basic registers with reset and clock which might not be used in synchronizers 

the way to solve this is to put naked registers in the syncs with a distinctive name so that the ASYNC REG property only catches those.

Hence, this issue seems mute.

AndreMerendeira, 2024-09-18:
Yes, that seems a better solution

closed 2024-09-18 by jjts